technologiescollege/Blockly-at-rduino #1: Pin entry now broken on most blocks

EricSKennedy, 2015-06-16:
Just updated my fork and now most of the blocks don't have a working way to enter digital or analog pins.
I see an update changed the way pins are being addressed.  Is there any way to make both the new way and the old drop-downs work?  (Sort of compatibility mode?)
Otherwise, someone needs to go fix all the pin references before this version is usable again.
Also I'd love to be able to use A0-5 as a digital pin, since it is physically possible on the hardware.  (I'd just figured out an update to make it work with the old dropdown pin method.)

SebCanet, 2015-06-16:
Well it was a hard decision because it breaks compatibility with BlocklyDuino. But it's the only way to keep it working with any Arduino card.mmmm.... we'll see if it's possible to have both.

What do you mean by 'needs to go fix all the pin references' ? what is the problem exactly ?

And for the use of A0-5 as digital pin, it's real problem for me because it's made to use with kids, and I'm not really sure that they will understand it quite well...I'll think of it....
Thanks.

EricSKennedy, 2015-06-16:
As far as A0-5, my kids age 12-7 understand it quite well, but I'd be fine with a switch somewhere that could be defaulted to off.

As to the inputs being broken, I tried again, and it looks like I just hit a few broken ones on my initial test.  I can't seem to enter Pin numbers in Grove-In-Temp Sensor, Joystick, or sound sensor.  Will do more testing later.

closed 2015-06-17 by EricSKennedy