technologiescollege/Blockly-at-rduino #5: Codebender plugin integration is broken

Arachnid, 2015-08-15:
Due to codebender's introduction of compilerflasher_new and this bug (https://github.com/codebendercc/compilerflasher/issues/1), codebender plugin integration is currently broken. Even once the bug is fixed, the code will need modifying to work with the new plugin script.

SebCanet, 2015-08-15:
Thanks for information. I alrdeay add this problem few weeks ago.
But I thought the problem was corrected ? When I test it works...
Tell me if you still have the pb.

Arachnid, 2015-08-15:
It definitely doesn't work for me - it tries to load a file that doesn't exist, and never enumerates the devices.

Arachnid, 2015-08-17:
I suspect this is a matter of what version of the plugin you have. Have you updated yours recently?

SebCanet, 2015-08-17:
Yes, and Blockly@rduino is normally updated.

Arachnid, 2015-08-17:
Perhaps it's a platform issue, then? I'm using Chrome on OSX.

If you look at codebender's own site, they have a "compilerflasher_new" which works with my version of the plugin, while "compilerflasher" that blockly@rduino is using does not. I tried porting blockly@rduino to use compilerflasher_new, but encountered the bug I linked to above.

SebCanet, 2015-08-20:
Normally it's platform independant, but the compilerflasher doesn't work with my Cyberfox x64. Do you have a 64bits version ?

Arachnid, 2015-09-06:
I do. If you look at the include code on codebender's site, it looks like this:

---
		                if (window.useChromeApp()) {
			                compilerflasher = new compilerflasher_new(function () {return window.files;});
			                compilerflasher.eventManager.addListener('plugin_not_found', function () {
				                window.location.assign("/static/walkthrough/page/2");
			                });
		                }
		                else {
			                compilerflasher = new compilerflasher(function () {return window.files;});
			                var pluginSearchedInterval = setInterval(function () {
				                if (compilerflasher.pluginHandler.plugin_searched) {
					                clearInterval(pluginSearchedInterval);

					                if (!compilerflasher.pluginHandler.plugin_found) {
						                window.location.assign("/static/walkthrough/page/2");
					                }
				                }
			                }, 100);
		                }
---

So it looks like the new plugin is specific to Chrome.

SebCanet, 2015-09-06:
As you can edit in index.html line 45, the reference is no more for '/utils/compilerflasher.js' but for 'http://codebender.cc/embed/compilerflasher.js'.

Did you try this one : https://codebender.cc/static/plugin ?

Arachnid, 2015-09-06:
Yes, the issue here is that even if you change the reference in index.html to compilerflasher_new, it still fails, due to the linked codebender bug.

SebCanet, 2015-09-10:
I do not understand because it works on my Windows Firefox x32 version...
Did you try witj all browser ? or on a different OS ?

Arachnid, 2015-09-11:
If you look at the snippet I pasted, compilerflasher_new is used for Chrome only. So as it stands, Blockly-at-rduino will not work in Chrome for any user with the latest Codebender plugin. I don't think there's anything we/you can do about it until Codebender fix their plugin, but I wanted to make you aware of the issue and how it can be fixed once they do.

closed 2015-12-25 by SebCanet