technologiescollege/Blockly-at-rduino #10: Traduction bluetooth block in french

jbet, 2016-04-02:
Le module Groove Bluetooth est mal traduit. 
Recevoir une trame bleutooth est correct. 
Mais Lire, est faux, il faut écrire Emettre vers bleutooth .... ou quelque chose comme ceci.

Voir ce même module en langue espagnol, qui lui est bien traduit.

Cordialement.

JBE

SebCanet, 2016-04-02:
C'est déjà fait : #9

closed 2016-04-02 by SebCanet

jbet, 2016-04-02:
Peut être, mais pas en ligne, car je ne l'ai pas trouvé.

SebCanet, 2016-04-02:
Si, si ,c'est sûr :-D
Le but de ce github est de stocker les fichiers, donc à l'utilisateur de télécharger les mises à jour.

jbet, 2016-04-02:
Alors il y a quelque chose que je n'ai pas compris.
Je voyais arrivé des mises à jour en allant sur le site. Très bien, le programme avance, et les petits problèmes vont pouvoir se résoudre.

Maintenant vous  m’annoncez qu'il faut télécharger le point zip, et l'extraire sur mon PC.
Ce n'est donc plus une application en ligne, **ou alors c'est un peu plus complexe**.

Si  vous avez des infos, je suis preneur, car j'ai vingt PC ou l'installer, et j'ai déjà fait la pub de votre logiciel à une trentaine de collègues , et déjà certain me demande comment l'utiliser.

Et à la fin du mois, il faut que je mettre en place pour mes collègues, prof de techno, API INVENTOR,  et votre logiciel. Cela ne va pas être simple pour quelque uns. C'est vrai que les objets connectés vont être un sacré défit pour nos collégiens...

jbet, 2016-04-02:
C'est bon cela fonction, en téléchargeant le zip, puis après l'avoir décompressé, cliquer sur index.html.
Mais j'aimerais quand même bien comprendre, est ce que l'élève va travailler à partir de sa session, à partir du poste mais en administrateur, ou  faudra il le mettre sur le serveur du collège.
Du travail en perspective.

SebCanet, 2016-04-03:
Téléchargement = local, donc si chaque éléève télécharge c'est dans sa session.
Si déposé sur un site web, autonome mais il faut faire les mises à jour en reprenant depuis le github.
Si en partage sur le serveur, chaque élève peut y accéder et il y a juste à en écraser les fichiers à chaque fois facile à mettre  jour. Je l'ai construis pour qu'il ne nécessite pas de réglage particulier, pas de serveur web, juste un partage de fichiers.