technologiescollege/Blockly-at-rduino #14: Firmata

jbet, 2016-04-04:
Dans Blockly@rduino, il y a un dossier Firmata.
Est ce Blockly@rduino est compatible avec Firmata !
Y a t il une doc pour savoir comment l'utiliser ?
J'ai vu comment installer Firmata+, sur youtube

A plus

SebCanet, 2016-04-04:
Il n'y a pas de dossier Firmata ??!
il n'y a pas de 'compatibilité' car ton programme remplace celui chargé dans la mémoire de la carte (ex : Firmata). Mais si tu téléverses FirmataPlus dans la carte, l'onglet supervision devient utilisable.

jbet, 2016-04-05:
Bien, merci pour la réponse sur Firmata, je vais essayer.

Pour info, car je pense que tu le sais, on ne peut plus accéder au site.

J'aurais une demande :
1 Implanté une possibilité #define..., je te précise demain ce que j'ai en tête...

Mais quand je vois les dates dépôts sur le site Github, il faut dormir de temps en temps, ou alors tu es très fort...

Bon courage.

----- Mail original -----

closed 2016-04-18 by SebCanet