technologiescollege/Blockly-at-rduino #15: Bloc define

jbet, 2016-04-06:
La définition d'une variable nous permet de racrrocher à une variable une instruction.
Exemple : #define LED_allumée digitalWrite( LED, High)
quand dans une action on mettra LED_allumée, le compilateur substitue LED_allumée par digitalWrite( LED, High). Et cela devient alors beaucoup plus facile pour des élèves grand débutants de commencer les algorithmes.
Pourriez vous faire, 2 blocks spécifiques, en s'inspirant de ce qui existe ;

Le premier pour des informations en entrées du microcontrôleurs :
[image]

Qui se lira
Définir BP1 comme étant l'état logique de la broche numérique ( numéro, ou nom de variable),
Tout ceci en un seul bloc.
La broche numérique sera alors mise en entrées ( INPUT)

Le second pour des informations en sortie du microcontrôleurs :

bmcage, 2016-05-12:
This is bad practice. In my view, kids must learn correct paradigms.

My approach you see here: http://ingegno.be/Manuals/Blockly4Arduino/blockly4Arduino/index_en.html?url=examples/Blink.xml

You can also change the generic digital output block, by a LED block. Then kids can just use a block: put LED on and put LED off. See this used in example Blink No Delay.
Note that coding with No Delay is easy with the effect block, but the code is then advanced coding level!

SebCanet, 2016-05-14:
I totally agree with you bmcage, I think it's confusing algorithm and program.
For younger novice, I created a multi level toolbox to choose what kind of details you allow, it can help for very new user but not really a practice to keep. It's not very difficult to make understandable 'function' utility.

closed 2016-05-14 by SebCanet