technologiescollege/Blockly-at-rduino #17: Soucis avec la dernière version

jbet, 2016-04-19:
J'ai téléchargé la dernière version à partir du Github, et voilà le résultat :
[image]

Il n'y a plus de correspondance entre les blocs et le code.

Bon courage.

SebCanet, 2016-04-24:
Avec le typag intégré (et obligatoire maintenant, ça marche) 
[image]

technoman72, 2016-04-25:
Manque la traduction de NUMBER en nombre ou entier ? 

-------- Message d'origine --------

  
  

Manque la traduction de NUMBER en nombre ou entier ?�
-------- Message d'origine --------
https://cloud.githubusercontent.com/assets/6362199/14771339/504049d8-0a88-11e6-9114-115ee2903a16.JPG https://cloud.githubusercontent.com/assets/6362199/14771339/504049d8-0a88-11e6-9114-115ee2903a16.JPG
https://github.com/notifications/beacon/ALcw4Dw5wPpib3EzbwSYbYPXw3p3tUcyks5p7AIsgaJpZM4IKXX3.gif

SebCanet, 2016-04-25:
trad faite mais pas automatique encore.

closed 2016-04-25 by SebCanet

jbet, 2016-04-25:
OK j'ai essayé, et cela fonctionne.

Mais cela aurait été mieux de pouvoir faire un #define , comme je l'ai mis sur le github.

Pour résoudre le problème d'algorigramme, algorithme au brevet, je vais leur apprendre les algorigrammes, et on fera la traduction en algorithme, avec votre logiciel.

Bonne soirée.

----- Mail original -----

capture