technologiescollege/Blockly-at-rduino #24: problème de typage avec une boucle "compter avec" (for)

PhilippeBrangier, 2016-05-13:
Bonjour, Je rencontre un problème avec la boucle for : la variable choisie est typée par défaut "void"

[image]

code généré : 
```
void i;

void setup() {
  pinMode(i, OUTPUT);
}

void loop() {
  for (i = 1; i <= 10; i=i+1) {
    digitalWrite(i, HIGH);
  }

}
```
ce code ne passe pas la vérification : 
```
Verification failed. Erreur : (sketch file) sketch.ino:1:6: error: variable has incomplete type 'void' void i;      ^
1 error generated.
```

Contournement : 
Je suis obligé de déclarer manuellement la variable ce qui me crée une alerte dans le bloc for : 
[image]

Le code généré passe la vérification : 
```
int i;

void setup() {
    i = (int)(0);

  pinMode(i, OUTPUT);
}

void loop() {
  for (i = 1; i <= 10; i=i+1) {
    digitalWrite(i, HIGH);
  }

}
```

Cordialement

Philippe

SebCanet, 2016-05-14:
Le typage que j'ai commencé à inotrduire pose des pb autant qu'il en résoud...je planche sur un système qui permet le typage de façon mieux intégrée, mais j'ai peu de temps pour ça...
Bref le meilleur est encore à venir ! :-D

Vraiment content que ça vous plaise et surtout c'est gagné si ça sert à des enfants ! C'est le but premier, je l'ai développé pour l'EducNat pour aussi pour des animateurs non spécialistes et l'éducation populaire : http://www.libreduc.net/

A ce titre je suis preneur pour le blog de tout article/idée de projet facile et pas cher à réaliser pour tous.

Merci.

closed 2016-05-14 by SebCanet

PhilippeBrangier, 2016-06-09:
Merci,
  En attendant, pour que mes fils n’aient pas à déclarer leurs variables, j'ai modifié \utils\blocklyDuino.js lignes 371-372 : <pre>    case Blockly.Types.NULL.typeName:return 'int';<\pre>