technologiescollege/Blockly-at-rduino #25: Problème d'affectation des pins lors de l'utilisation d'une variable avec le bloc "mettre la broche numérique""

PhilippeBrangier, 2016-05-13:
Bonjour ,

 Je rencontre un problème avec le bloc "mettre la broche numérique ... à l'état logique" lorsque je l'utilise avec une variable pour le numéro de broche (tous les blocs de ce type sont susceptibles d'êtres sujets à ce problème) : 
[image]
le code généré contient "pinMode(MaVariable, OUTPUT);" : 
```
void i;

void setup() {
  pinMode(i, OUTPUT);
}

void loop() {
  for (i = 1; i <= 10; i=i+1) {
    digitalWrite(i, HIGH);
  }

}
```
Mes pins de sortie ne sont donc pas mise en mode "output"

Contournement (j'utilise deux variables pour mettre en lumière les problèmes rencontrés) : 
[image]
qui me génère : 
```
int i;
int j;

void setup() {
    i = (int)(1);
  j = (int)(0);
  for (j = 1; j <= 10; j=j+1) {
    pinMode(j, OUTPUT);
  }

  pinMode(i, OUTPUT);
}

void loop() {
  for (i = 1; i <= 10; i=i+1) {
    digitalWrite(i, HIGH);
  }

}
```
Il faut bien penser a initialiser la variable utilisée par le bloc a la valeur d'une des pins utilisée en sortie sinon elle vaut 0 et c'est donc la pin 0 qui est passée a "output"

Je n'ai pas trouvé de bloc pour l'affectation des pins, je suis donc passé par un bloc "ligne de code"

suggestions : 
- Ce bloc ne devrait pas tenter d'affecter des pins si l'argument est une variable 
- Un bloc permettant l'affectation des pins pourrait être créé pour éviter l'utilisation du bloc "ligne de code"

Cordialement 

Philippe

SebCanet, 2016-08-20:
Un bloc permettant l'affectation des pins pourrait être créé pour éviter l'utilisation du bloc "ligne de code" me semble une bonne idée, à rajouter donc...

SebCanet, 2018-01-30:
Fixé....merci.
[image]

closed 2018-01-30 by SebCanet