technologiescollege/Blockly-at-rduino #28: Ordre dans l'initialisation des variables

cdurski, 2016-06-27:
Bonjour,

tout d'abord, bravo pour votre travail. Nous pensons utiliser Blocky@rduino en option ISN pour l'introduction à l'algorithmique sans passer par l'apprentissage trop abrupt du C pour Arduino.

En testant, nous sommes tomber sur un "os" dans le bloc setup.
[image]
Ce qui donne le code suivant
`int etatBouton;
int LED;
int bouton;

void setup() {
  pinMode(bouton, INPUT);
  pinMode(LED, OUTPUT);
    etatBouton = (int)(HIGH);
  LED = (int)(8);
  bouton = (int)(2);}
void loop() {
    etatBouton = (int)(digitalRead(bouton));
    if (etatBouton == HIGH) {
      digitalWrite(LED, HIGH);    } else {
      digitalWrite(LED, LOW);
    }}

Par défaut la déclaration des variables se fait après la déclaration des pinMode. ce qui fait qu'en fait les pins ne sont pas reconnus par leurs noms.

Nous avons contourné le problème en insérant une ligne de code manuelle.
Une suggestion pour que ce soit plus propre (avec une déclaration de variable en dehors du bloc setup par exemple mais pas dans le bloc loop comme on le fait en programmation "normale" ?`

SebCanet, 2016-07-08:
Merci, il y a un vrai souci avec ça....
La seule solution possible pour l'instant, c'est à dire avant que je n'ai le temps de me plonger dedans, est de passer par le bloc "Dfinitions" ainsi que l'accolade qui va avec, c'est ce que j'ai pu faire sans modifier le comportement par défaut de Blockly : 
[image]

Qu'en pensez vous ? Tout est modifiable, le texte, le comportement (dans une moindre mesure :-D )

cdurski, 2016-07-09:
Très bien ! Nous n'y avions pas pensé.....
Merci beaucoup ! Nous vous tiendrons au courant de l'intérêt des élèves pour cette méthode

SebCanet, 2016-08-20:
Il faut que je créé un bloc qui fait la déclaration + l'attribution d'une valeur à une variable dans une même action. A venir.

cdurski, 2016-10-31:
Nous venons de tester avec nos élèves d'ISN qui, cette année, étaient pour la plupart tous novices en programmation et en électronique. 
l'utilisation de Blockly@rduino  nous a grandement facilité la tâche et a rendu les séances beaucoup plus arbodables d'un point de vue pratique et d'un point de vue théorique Merci pour votre travail.

Je reviens sur le problème de déclaration de variable. j'avais testé chez moi en fin d'été et tout fonctionnait très bien.  Malheureusement au moment de le mettre en application avec les élèves message d'erreur ci-dessous. Je vous soumets donc le problème car je ne vois pas ce que sont les "identifiers" demandés.

[image]

Message d'erreur
[image]

cdurski, 2016-11-01:
Bonjour, je viens de me repencher sur le problème et ai trouvé une solution 👍 

[image]
 Et ça fonctionne !

Je ne m'explique toujours pas le message d'erreur précédent. Si à l'occasion vous pouviez me répondre....

Merci d'avance et encore bravo pour votre travail !!!!

SebCanet, 2016-11-01:
Bonjour,
l'ancien problème venait de la solution de typage qui surhacrgeait l'information. Avec la 'mécanique' actuelle c'est résolu.
[image]

closed 2016-11-01 by SebCanet