technologiescollege/Blockly-at-rduino #43: Fermeture de codebender.cc

jm4r7in, 2016-10-20:
[Codebender ayant annoncé sa fermeture](https://codebender.cc/next-chapter), quelles sont les solutions envisageables ?
Je peux envisager de mettre un serveur à disposition, mais j'aurais besoin d'aide pour la mise en place...

[Codebender is closing](https://codebender.cc/next-chapter), what solutions can we imagine ?
I can use a server, but I'd need some help to set it...

SebCanet, 2016-10-21:
Merci de cette proposition ! C'est très généreux. Je n'ai aucune idée du type de serveur à mettre en place pour remplacer le service Codebender. Même si les sources sont sur Github, le projet intègre trop de choses. Et il faut voir à l'usag car ils gardent des parties utilisables: Blocks + Edu. Donc le plugin va peut être pouvoir continuer à être utilisable...

Conscient des craintes du 'tout en ligne' (pérennité, contrôle, gratuité, etc), je vais continuer à travailler sur ma version 'tout en un' hors ligne.

Question sur Codebender à laquelle je n'ai aucune réponse : pourquoi leur compilation est aussi rapide ???

jm4r7in, 2016-10-21:
As-tu une solution pour compiler hors-ligne ?

Ils compilent peut-être avec un compilateur plus rapide, ou ils ont peut-être des serveurs puissants...

SebCanet, 2016-10-22:
Pour compiler hors ligne, j'ai:
[Blockly@rduino-communication](https://github.com/technologiescollege/Blockly-rduino-communication) en complément au site web
[Blockly@rduino_AIO](https://github.com/technologiescollege/Blockly-rduino_AIO) pour tout hors ligne, avec l'EDI, le navigateur, etc

J'attends de voir s'ils ferment complètement, auquel cas je vais trancher dans Blockly@rduino.

Je pense que le pb de vitesse de compilation n'est pas lié que à la machine mais aussi à l'OS.

jm4r7in, 2016-10-25:
Félicitations pour ces 2 solutions.
En attendant d'avoir un compilateur écrit en JavaScript... (as-tu une idée si c'est faisable ?)
Sinon, j'aime quand même bien l'idée de compiler sur un serveur en ligne...

SebCanet, 2016-10-25:
Pour le compilateur, je manque de documentations pour savoir si on peut taper dans l'éditeur web d'Arduino Create en remplacement de Codebender...
On pourrait imaginer lui envoyer le sketch en argument, mais je n'ai pas pris le temps de commencer mes recherches là-dessus. Trop de boulot sur le hors ligne pour l'instant.
Du coup je tag 'besoin d'aide' :-D

SebCanet, 2017-05-06:
Blockly@rduino_AIO étant fonctionnel, je pense qu'on peut clore ce sujet.

closed 2017-05-06 by SebCanet