technologiescollege/Blockly-at-rduino #46: SCOOP librairie

SebCanet, 2016-11-06:
or anything to parallel tasks.
ou n'importe quelle bibliothèque pour des taches en parallèle.

[https://github.com/pietern/avr-tasks](https://github.com/pietern/avr-tasks)

philippejadin, 2018-12-17:
Est-ce toujours d'actualité ?

Je verrais bien quelque chose comme ceci : 

- blocs "Action planifiée"
- le bloc permet de planifier l'exécution d'une action. Toutes les xx secondes/millisecondes
- on peut y mettre ce qu'on veut (comme dans une boucle)

en arrière boutique, chaque bloc crée une fonction qui utilise le timer pour voir si elle doit s'éxécuter
dans loop() on appelle chaque fonction planifiée tout le temps

Je pense pouvoir faire ça, ça me fera un bon exercice

SebCanet, 2018-12-17:
Bonjour,
oui oui tout ce que je n'ai pas le temps de faire est toujours d'actualité...
J'essaierai de mettre à jour le wiki pour créer des blocs pour améliorer les captures mais le principe n'a pas changé.
Suis dispo par mail ou sur le forum y'a du monde pour dépanner.
Merci.