technologiescollege/Blockly-at-rduino #48: add 28BYJ-48 Stepper Motor with ULN2003 driver

SebCanet, 2016-11-14:
References:
[http://42bots.com/tutorials/28byj-48-stepper-motor-with-uln2003-driver-and-arduino-uno/](http://42bots.com/tutorials/28byj-48-stepper-motor-with-uln2003-driver-and-arduino-uno/)
[http://tiptopboards.free.fr/arduino_forum/viewtopic.php?f=2&t=41](http://tiptopboards.free.fr/arduino_forum/viewtopic.php?f=2&t=41)

openelearning, 2018-12-27:
à voir donc si je pourrai ajouter un bloc pour ce moteur pas à pas. Dans https://github.com/technologiescollege/Blockly-at-rduino/issues/141#issuecomment-450011215 tu indiques "je veux bien que tu me donnes une référence + lien pour le helpUrl + photo."
Je vais donc essayer de les appliquer pour ici.
1) Il y a déjà 3 références ci-dessus, ce n'est donc plus la peine, non ?
2) un lien pour le help url : désolé je ne sais pas ce que sait (je débarque dans blockly at arduino :) ) tu sais s'il y a des exemples d'help url pour que je vois ce que c'est (un tuto sur l'utilisation du bloc ?)  ?
3) pour la photo ok, j'en prendrai ou j'en trouverai compatible avec la GNU/GPL.

openelearning, 2018-12-29:
1. En fait c'est le ULN2003A, je ne sais s'il y a une différence avec ULN2003.
3. Oui pas évident la gestion des licences... Pas trouvé de photos sous licence libre, mais j'ai pu trouvé pour le moment un bel SVG représentatif issu du domaine public (openclipart) dont j'ai mentionné la licence et le lien dans le code (en cours d'écriture).

SebCanet, 2019-01-02:
Zou c'est bouclé ! Merci.

SebCanet, 2019-01-13:
Still few problems...

tutoblockly, 2019-01-21:
Quand on colle les 2 blocs de déclaration, il y a un code qui disparaît.

tutoblockly, 2019-02-01:
Problème corrigé.

SebCanet, 2019-02-01:
C'est bouclé grâce à ta mise à jour d'hier ?

SebCanet, 2019-02-01:
Super, merci ! On cloture.

closed 2019-02-01 by SebCanet