technologiescollege/Blockly-at-rduino #58: Pertes des items dans le menu Arduino

jmchiappa, 2017-03-04:
Bonjour,
je viens de télécharger la dernière version 24-02-2017 - v2.3 "Kit Cat"et les blocks dans les sous-menus "Arduino/entrées", "Arduino/Communication série" et "Arduino/communication logicielle" ont disparu.

Dans la console Javascript, j'ai l'erreur suivante lorsque je clique sur l'un des sous-menus :
`TypeError: c is undefined[En savoir plus] blockly_compressed.js:1322:254

Je n'ai plus non plus le dropbox pour définir la langue dans l'interface principale.

Je suis sous Firefox 51.0.1

Dites-moi quelles sont les informations utiles pour vous et merci d'avance pour votre aide.

SebCanet, 2017-03-12:
Bonjour,
pour bien vérifier le comportement de chaque version, il faut bien nettoyer tous les caches du navigateur ainsi qu'en cliquant sur le logo en haut à gauche.

Merci de me dire si cela fonctionne.

jmchiappa, 2017-03-12:
Bonjour,
Je viens d'essayer en vidant le cache firefox (maintenant 52.0).
j'ai le même résultat. Voici un snapshot en espérant que cela peut vous aider.
[image]

SebCanet, 2017-03-12:
C'est dû à la version que tu as téléchargée qui doit contenir un bug.
Teste directement ici: [http://technologiescollege.github.io/Blockly-at-rduino/](http://technologiescollege.github.io/Blockly-at-rduino/)

jmchiappa, 2017-03-12:
Même constat...
[image]

SebCanet, 2017-03-12:
C'est parce que tu n'as sélectionné aucune carte ?

jmchiappa, 2017-03-12:
ça a l'air de marcher (je me demandais pourquoi le p'tit bonhomme se posait des questions...). Merci et bonne soirée !
je le saurais à l'avenir ;-) !

closed 2017-03-12 by jmchiappa