technologiescollege/Blockly-at-rduino #79: i can't found blocks !!!

hamam1, 2017-09-29:
[image]

SebCanet, 2017-09-29:
First you choose your card, then select your level of 'difficulty', from 0 to ALL.
You click on "configure blocks", blocks from difficulty level.
That's all.

hamam1, 2017-09-29:
Thank you very much <3 <3 <3 <3 <3 
please can you help me how i can use "arduino  ir remote control "
and how i can make new block 
sincerely <3 <3 <3 <3 <3

SebCanet, 2017-09-29:
For the arduino ir remote control, I don't know because I'm not the creator of this blocks....
To create block, you have french doc there: [http://info.technologiescollege.fr/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling](http://info.technologiescollege.fr/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling)
and english version (but not updated...if you want to help me !?): [http://info.technologiescollege.fr/wiki/doku.php/en/blockly_rduino/create_blocks](http://info.technologiescollege.fr/wiki/doku.php/en/blockly_rduino/create_blocks)

Hope it can help.

hamam1, 2017-09-29:
there are any vedio  can help me to create a new block 
becuse i am from egypt and iam not good in english <3

SebCanet, 2017-09-29:
I never made video to explain, it would be too much long. And it would be with many comments (in French or english), so you will have the same pb. Maybe you could try by online translation.

closed 2017-10-18 by SebCanet