technologiescollege/Blockly-at-rduino #140: Ajouter capteur de particule fine

philippejadin, 2018-12-17:
Pour le boulot, nous allons réaliser des stations météo et détection de particule fine. Un capteur utilisé est le SDS011 ( utilisé par exemple par https://luftdaten.info )

Je ferais bien un bloc pour ce capteur. Sera-t-il intégré dans la version officielle si je fais un pull request?

SebCanet, 2018-12-17:
Je prends tout ! Aucun souci, mon objectif est que tout serv à tout le monde, libre à chacun d'enlever par la suite.
Merci d'avance !

philippejadin, 2018-12-25:
Hello,

Je travaille à avoir un format de bloc "standardisé" du cou le premier bloc que j'ai fait est celui pour le SDS011
Il est ici : https://github.com/philippejadin/blockly-arduino/tree/master/blocks/sds011-particle-sensor

L'idée est de pouvoir simplement ajouter ce dossier à une install blockly, et qu'elle ajoute le block à la liste des blockes disponibles.

Si on se met d'accord sur un format, je m'engage à convertir les blocs blockly@rduino les plus utiles sur ce canevas.

SebCanet, 2018-12-30:
Ce qui me vient à l'esprit rapido:
- mettre la couleur en relatif, voire comme j'ai fait en global pour faciliter 
- nommer dossier et fichier pour mettre en correspondance, afin de faciliter le debuggage
- reprendre la syntaxe d'écriture officielle Blockly: Blockly.Blocks['sds011 ']