technologiescollege/Blockly-at-rduino #141: Corrections pour le moteur pas à pas ?

openelearning, 2018-12-26:
Bonjour,

Tout d'abord, merci pour ce superbe outil !

Je débute dans blocly-at-arduino, dans les moteurs pas à pas, et ne maîtrise pas js. Je ne suis donc pas sûr de ce qui suit.

Il semble que le bloc moteur pas à pas ne fonctionne pas très bien. Pour broche 1 et broche 2, si on met par exemple 8 et 9 pour, dans le code arduino qui en sort, on a "null".
En revanche si on va dans generators/arduino/stepper_motor.js et que : 
On supprime : 
var pin1 = block.getFieldValue('STEPPER_PIN1');
var pin2 = block.getFieldValue('STEPPER_PIN2');
On ajoute : 
var pin1 = Blockly.Arduino.valueToCode(block, 'STEPPER_PIN1',
      Blockly.Arduino.ORDER_ATOMIC) || '8';
var pin2 = Blockly.Arduino.valueToCode(block, 'STEPPER_PIN2',
      Blockly.Arduino.ORDER_ATOMIC) || '9'; 
Cela semble fonctionner.

Est-ce que quelqu'un a le même fonctionnement, et si oui, est-ce que cette proposition de correction semble envisageable ?
Merci

SebCanet, 2018-12-26:
Bonjour,
ce n'est pas étonnant qu'il y ait des bugs vu le nb d'éléments que je ne peux pas tester.
Si tu as le matériel et que tu me confirmes que tes modifs fonctionnent, pas de souci tu me proposes un push ou je rajoute, pas de souci. Dans la foulée je veux bien que tu me donnes une référence + lien pour le helpUrl + photo.
Merci !

openelearning, 2018-12-27:
Salut,
J'ai pas encore bien saisi ce bloc, il ne demande que 2 pins, alors que le moteur pas à pas que j'ai correspond plus à https://github.com/technologiescollege/Blockly-at-rduino/issues/48 où il faut 4 pins. (Mais il m'a semblé avoir vu sur le net qu'on pouvait gérer des moteurs pas à pas avec seulement 2 pins).

Par prudence on a qu'a laissé comme tel pour ce bloc déjà inclus dans blocky@rduino (sauf si ça te saute aux yeux que c'est un bug dans la modif proposée) et de mon côté je vais essayer (à voir si j'y arrive) de proposer un nouveau bloc de moteur pas à pas nécessitant 4 pins et que je pourrai tester. à bientôt et merci pour ton retour :)