technologiescollege/Blockly-at-rduino #142: Documentation : contenu d'une page en double

openelearning, 2018-12-28:
Super cette doc, milles merci !

On dirait que sur http://libreduc.cc/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling/blocdefext , il y a une duplication du contenu : Après "Ouf c'est fini !" ça recommence du presque début.

SebCanet, 2018-12-29:
Merci pour la relecture. Le changement de domaine semble avoir modifié les fichiers...
Merci de me dire s'il y a des éléments à éclaircir, mettre à jour, etc. Je peux rajouter des utilisateurs pour cexu qui veulent contribuer.... :-D

closed 2018-12-29 by SebCanet

openelearning, 2018-12-29:
Merci pour le retour. Ok n'hésites pas à m'ajouter au cas où je vois d'autres petites corrections à faire (pour les corrections plus importantes et celles dont je ne suis pas sûr, je reviendrai vers toi avant).