technologiescollege/Blockly-at-rduino #153: Adding Variables creates errors

adamdrenth, 2019-05-13:
I am getting the message "<some variable> has already been added" over and over after I create more than two variables. The variables do have simple names: btnAState, btnBState, playerAcounter, playerBcounter and gameOn.  Adding the last three variables created the errors.

SebCanet, 2019-05-13:
Hi,
you're right, there's a pb with variable creation.
I do not have time for few weeks, but I'll try to check asap.
Thanks.

adamdrenth, 2019-05-14:
That would be great. It seems like the problem is exponential. By the time I've added 5 variables or more, I have to click "okay" over 100 times.

SSoulard, 2019-11-30:
Bonjour,
Tout d'abord, bravo pour votre travail. L'outil est vraiment idéal avec les élèves, surtout depuis l'intégration dans l'IDE.

Je reviens malgré tout sur ce problème à la création des variables qui bloque plusieurs de nos collégiens. 
Existe-t-il une alternative pour empêcher l'apparition de ce message ? En effet, ils n'osent plus créer plus de 5 variables afin d'éviter d'avoir à cliquer indéfiniment dans la boite de dialogue.

Merci d'avance.

SebCanet, 2019-12-01:
Bonjour,
je suis en train de travailler sur un reboot du noyau de Blockly@rduino afin de bénéficier des nouveautés du framework Blockly, et dedans le pb des variables sera corrigé...mais je n'avance pas vite car étant prof à plus que plein temps je n'ai pas beaucoup de temps à y consacrer hélas.

Dans Firefox il faut qu'il coche un truc qui doit écrire "ne plus recevoir de message de la page" ou quelque chose du genre....

Ce n'est absolument pas satisfaisant hélas, mais je ne peux rien offrir de mieux pour le moment.

Bref, merci de la patience....

SebCanet, 2019-12-31:
Problème fixé.

closed 2019-12-31 by SebCanet

SSoulard, 2020-01-01:
Merci beaucoup pour votre réactivité !
Je vais aller tester ça de suite.

Bonne fin de "vacances" et meilleurs voeux.