technologiescollege/Blockly-at-rduino #166: Dropdown image

LcKoh0830, 2020-03-20:
Hi, is that possible we can add image into the dropdown option in block?

SebCanet, 2020-03-20:
Hi,
it's possible, you have example in Blockly's official guide: [https://developers.google.com/blockly/guides/create-custom-blocks/fields/built-in-fields/dropdown#image_dropdowns](https://developers.google.com/blockly/guides/create-custom-blocks/fields/built-in-fields/dropdown#image_dropdowns)

Enjoy.

LcKoh0830, 2020-03-20:
@SebCanet Hi, thanks for the answer. Is it possible we can add the function (dropdown with image) into Blockly@duino? If yes, can you tell me how?

SebCanet, 2020-09-14:
@LcKoh0830 Hi,
really sorry, I discover just now your answer...
Where would you add this function?

LcKoh0830, 2020-09-15:
Hi, I had already forgot where should the function added.
But I think is the field_dropdown function do not have the function to handle image.
Blockly's official field_dropdown.js have the function to handle image, the blockly js here had been compressed, so I am not sure how to add it.

SebCanet, 2020-09-15:
Maybe it will be a function for the new BlocklyDuino: [https://github.com/BlocklyDuino/BlocklyDuino-v2](https://github.com/BlocklyDuino/BlocklyDuino-v2)

closed 2021-05-04 by SebCanet