technologiescollege/Blockly-at-rduino #171: Add the Time of Flight distance sensor

julien3108, 2021-04-28:
Hi

I've recently replaced the big ultrasonic SR04 distance sensor by the tiny VL53L1X Time of Flight distance sensor on a standard commercial robot car for obstacle avoiding purpose.
Please checkout the video here : https://photos.app.goo.gl/Hzf7tmJtGQtwfaPUA.
I did it in Arduino C code. I would like to create the blocks for this sensor for the children but I don't have any experience on the blocks creation. Would you be interested to work with me to make it?

Thanks
Julien

SebCanet, 2021-04-28:
Hello,
the library and the corresponding blocks are already described in Blockly@rduino, section Grove -> In -> VL53L0X
If you want to add blocks or improve the existing ones, the wiki is here: [http://technologiescollege.fr/wiki/doku.php](http://technologiescollege.fr/wiki/doku.php)
The English section is a bit outdated, but the French section should be usable.
If you have any problems, you can post on the forum: [http://blockly.technologiescollege.fr/forum/](http://blockly.technologiescollege.fr/forum/)
Enjoy tests.

julien3108, 2021-04-28:
Bonjour Seb, le premier lien que tu m'as envoyé ne fonctionne pas. Il dit
"cette page n existe pas encore , blabla" . Merci. Julien

SebCanet, 2021-04-28:
Et tu as le choix de la langue à droite, ça donne directement ça : [http://technologiescollege.fr/wiki/doku.php?id=fr:accueil](http://technologiescollege.fr/wiki/doku.php?id=fr:accueil)

SebCanet, 2021-05-04:
Sensor added: Grove -> In -> Distance time of flight VL53L0X

closed 2021-05-04 by SebCanet

julien3108, 2021-05-04:
Merci Seb
Pensez vous organiser une presentation/formation en ligne pour apprendre
aux gens de creer un block ? Je serai preneur. Julien

SebCanet, 2021-05-04:
Volontiers, on peut s'organiser via la liste de Pagestec pour avoir plus de monde ? Ou bien une autre liste de diffusion pour que ça bénéficie à un maximum de personnes.

julien3108, 2021-05-05:
Bonjour Seb, je te laisse communiquer ? Merci. Julien